jrester/tesla_powerwall #1: authentication to Powerwall software version 1.40.2

swm11, 2019-10-13:
Having read your excellent code and vloschiavo's documenation, I'm still unable to login to my Powerwall 2 running version 1.40.2.  vloschiavo note on authentication says "*** This documentation is old (created on version 1.15) and needs updating *** *** I wouldn't be surprised if less than 1% of the below still works in versions 1.40+ ***" - have you managed to authenticate with a new version 1.40+ ?
I can successfully login as a customer and and installer.

jrester, 2019-10-13:
Sorry, but I am currently unable to test the API. But could you further specify the problem? What are error messages do you see, when trying to login?

swm11, 2019-10-13:
> 
> Sorry, but I am currently unable to test the API. But could you further specify the problem? What are error messages do you see, when trying to login?

I modified your code to include a username since I believe that it is required to login to the system.  Login with the key on the gateway only seems to be possible after toggling the power switch (i.e. to ensure physical access to the system).

The error message I’m getting (with email and password edited):

DEBUG:tesla_powerwall_swm:Getting https://192.168.1.201/api/operation
DEBUG:tesla_powerwall_swm:Authenticating
DEBUG:tesla_powerwall_swm:Post {'username': ‘<email>', 'password': ‘<password', 'force_sm_off': True} to https://192.168.1.201/api/login/Basic
Traceback (most recent call last):
  File "./setBackupBatPc.py", line 15, in <module>
    print("mode=%s"%(power_wall.mode))
  File "/Users/swm11/Projects/homeautomation/powerwall/tesla_powerwall_swm.py", line 210, in mode
    return self.operation["mode"]
  File "/Users/swm11/Projects/homeautomation/powerwall/tesla_powerwall_swm.py", line 206, in operation
    return self._get("api/operation", True)
  File "/Users/swm11/Projects/homeautomation/powerwall/tesla_powerwall_swm.py", line 85, in _get
    self.login()
  File "/Users/swm11/Projects/homeautomation/powerwall/tesla_powerwall_swm.py", line 108, in login
    response = self._post('api/login/Basic', {'username': ‘<email>', 'password': self.password, 'force_sm_off': True})
  File "/Users/swm11/Projects/homeautomation/powerwall/tesla_powerwall_swm.py", line 100, in _post
    return self._process_response(response)
  File "/Users/swm11/Projects/homeautomation/powerwall/tesla_powerwall_swm.py", line 64, in _process_response
    raise InvalidPassword()
tesla_powerwall_swm.InvalidPassword: Invalid password

This is on a Powerwall 2 with version 1.40.2 firmware.

Thanks for your time - much appreciated.

- Simon

[Simon W Moore][http://www.cl.cam.ac.uk/users/swm11 ]

jrester, 2019-10-14:
I will take a look at this issue as soon as I have access to a Power Wall 2 again. If you like you can try to investigate this issue further. Maybe by intercepting the traffic of the Wizzard to see if the API has changed.

bdraco, 2020-02-26:
@swm11 The login endpoint has changed.  Its now

Old payload
```
{"username": "", "password": self.password, "force_sm_off": True}
```

New payloads
```
{"username":"installer","email":"<any_valid_email>","password":"<installer_password>","force_sm_off":false}
```
```
{"username":"customer","email":"<any_valid_email>","password":"<customer_password>","force_sm_off":false}
```

jrester, 2020-03-30:
@bdraco thanks for investigating the payload!

Should be fixed in version 0.1.3

closed 2020-04-09 by jrester