jrester/tesla_powerwall #5: API fails to initialize with version 1.46.0

ocalvo, 2020-04-17:
[More info](https://github.com/home-assistant/core/issues/34207#issuecomment-615437725)

jrester, 2020-04-18:
Would you mind sharing some responses from the API and the main javascript code? So I can include them I the API? The most important ones for homeassistant would be:
* `https://<powerwall-api>/api/site_info`
* `https://<powerwall-api>/api/system_status/soe`
* `https://<powerwall-api>/api/meters/aggregates`
* `https://<powerwall-api>/api/powerwalls`
* `https://<powerwall-api>/api/sitemaster`

If you feel like the response includes sensitive please consider replacing them with `XXX`.

And the main javascript code. This should be a file named something like `app.XXX.js`. In chrome you can find it in the developer console under "Sources" and in Firefox under "Debugger".

Thanks for helping!

ocalvo, 2020-04-18:
- https://tesla-gateway/api/site_info
```json
{"max_site_meter_power_kW":1000000000,
"min_site_meter_power_kW":-1000000000,
"nominal_system_energy_kWh":13.5,
"nominal_system_power_kW":15,
"max_system_energy_kWh":0,
"max_system_power_kW":0,
"site_name":"Home Energy Gateway",
"timezone":"America/Los_Angeles",
"grid_code":"60Hz_240V_s_IEEE1547a_2014",
"grid_voltage_setting":240,
"grid_freq_setting":60,
"grid_phase_setting":"Split",
"country":"United States",
"state":"Washington",
"distributor":"*",
"utility":"Puget Sound Energy",
"retailer":"*",
"region":"IEEE1547a:2014"}
```
- https://tesla-gateway/api/system_status/soe
```json
{"percentage":60.646753185128944}
```
- https://tesla-gateway/api/meters/aggregates
```json
{"site":
  {"last_communication_time":"2020-04-18T12:08:57.187708738-07:00",
  "instant_power":-5.4799957275390625,
  "instant_reactive_power":-177.02999877929688,
  "instant_apparent_power":177.11479560152986,
  "frequency":60,
  "energy_exported":468746.19806244376,
  "energy_imported":487472.18806244386,
  "instant_average_voltage":121.69499969482422,
  "instant_total_current":0,
  "i_a_current":0,
  "i_b_current":0,
  "i_c_current":0,
  "timeout":1500000000},
"battery":
  {"last_communication_time":"2020-04-18T12:08:57.19035638-07:00",
  "instant_power":-2900,
  "instant_reactive_power":20,
  "instant_apparent_power":2900.068964697219,
  "frequency":59.979,
  "energy_exported":679270,
  "energy_imported":817590,
  "instant_average_voltage":243.53333333333333,
  "instant_total_current":63,
  "i_a_current":0,
  "i_b_current":0,
  "i_c_current":0,
  "timeout":1500000000},
"load":
  {"last_communication_time":"2020-04-18T12:08:57.187708738-07:00",
  "instant_power":1135.123687456975,
  "instant_reactive_power":-207.97783434945234,
  "instant_apparent_power":1154.0193089401098,
  "frequency":60,
  "energy_exported":0,
  "energy_imported":1795053.746111111,
  "instant_average_voltage":121.69499969482422,
  "instant_total_current":9.327611572402613,
  "i_a_current":0,
  "i_b_current":0,
  "i_c_current":0,
  "timeout":1500000000},
"solar":
[... 361 lines cut ...]
                    "start_time": null,
                    "end_time": null,
                    "progress": 0,
                    "results": null,
                    "debug": null
                },
                {
                    "name": "Grid code",
                    "status": "not_run",
                    "start_time": null,
                    "end_time": null,
                    "progress": 0,
                    "results": null,
                    "debug": null
                }
            ]
        }
    },
    "states": []
}
```
- https://tesla-gateway/api/sitemaster
```json
{"status":"StatusUp","running":true,"connected_to_tesla":true}
```

ocalvo, 2020-04-18:
Here is the app.XXXX.js:

jrester, 2020-04-19:
Great thanks! That helps alot!

mago-kamar, 2020-04-22:
Hello,
sorry but I'm lost looking from my previous opened issue and this new one.

The fix has been released?

Thanks a lot

jrester, 2020-04-22:
@mago-kamar The fix for the unique_ids problem should be integrated in version Homeassistant 0.108.8. 
I am currently working on adding support for powerwall version 1.46.0 to Homeassistant.

mago-kamar, 2020-04-22:
> @mago-kamar The fix for the unique_ids problem should be integrated in Homeassistant as of version 0.108.8.
> I am currently working on adding support for powerwall version 1.46.0 to Homeassistant.

Perfect @jrester ,
thank you very much

mago-kamar, 2020-04-22:
Hi @jrester ,
I installed the 0.108.8, but the issue is still present.
I uninstalled the integration, and then I installed it again, but it doesn't work ... I have the same problem.

Thanks,

Enrico

jrester, 2020-04-22:
You mean that no entities show up?

mago-kamar, 2020-04-22:
> You mean that no entities show up?

Exactly :(

ocalvo, 2020-04-22:
> > You mean that no entities show up?
> 
> Exactly :(

Can you browse to your Tesla gateway website? What version do you see in the bottom?

mago-kamar, 2020-04-22:
1.45.2

jrester, 2020-04-22:
Could you reopen the other issue and post your logs there?

mago-kamar, 2020-04-22:
> Could you reopen the other issue and post your logs there?

ok

jrester, 2020-04-23:
Should be closed by [34545](https://github.com/home-assistant/core/pull/34545).

closed 2020-04-23 by jrester