jrester/tesla_powerwall #6: Running on Raspberry Pi;

omaliphant, 2020-04-25:
Hi,
Am using python on a raspberry pi to pull data from my powerwall and send it to my Crestron system.
I have an issue when running my script that errors on the __init__.py file.

Line 48; self.endpoint = f”https://{endpoint}”
Syntax error: invalid syntax

Have you seen this before? 
The Pi 3b+ is running Raspbian 9.4
Python is 3.5.3
Works great on my dev env in windows 10 / VS Code.
Many thanks

jrester, 2020-04-25:
Hey, python 3.5 does not support f'strings. Can you switch to a newer python version?  Otherwise I can implement a fix to make it compatible with older python versions.

omaliphant, 2020-04-25:
I will update to 3.7.4 and try again. Thanks for the quick reply!

omaliphant, 2020-04-25:
it seems Raspbian Strech doesnt support greater than 3.5... and i cant go to Buster as its got compatibility issues with the bluetooth integration that i am using on another app...
any advice?

jrester, 2020-04-25:
You could Install python fromm a different source but this isn't a good solution. I will try to make it more backwards compatible. I think I will publish a fix tomorrow.
Yes, a Note would be a good idea😃

omaliphant, 2020-04-25:
Thank you so much!

jrester, 2020-04-26:
I pushed a fix to master. Could you verify it works by cloning the repo and trying on your raspberry pi? Then I can publish a new version to PyPi.

omaliphant, 2020-04-26:
testing now... thanks!

omaliphant, 2020-04-26:
was also missing the pip "packaging" dependancy, added it with "pip3 install packaging" and now it works great! thank you so much!

closed 2020-04-26 by omaliphant

jrester, 2020-04-26:
Oh, ups. Seems Like I forgot to add it to the requirements.txt😬