jrester/tesla_powerwall #7: get_site_info() fails when there is no "region" in response

LordShadowen, 2020-05-18:
Trying the latest version of HA, the integration failed immediately upon start (both in YAML and through the UI). No helpful logging or error anywhere....

Tried a simple test python script with this library directly and this worked fine:

power_wall = Powerwall([IP address])
power_wall.get_site_info()

Looking through the code, I can see the setup sequence in HA is:

power_wall = Powerwall([IP address])
power_wall.detect_and_pin_version()
power_wall.get_site_info()

In my script this fails with error:

tesla_powerwall.error.APIChangedError: Powerwall api error: It seems like the Powerwall API changed for '<class 'tesla_powerwall.responses.SiteInfoResponse'>'': Some attributes where removed from the response: ['region']

Looking at the response of /api/site_info on my powerwall I see there is indeed no "region". 

{ [lots of stuff], "country":"United States","state":"California"}

So it seems the library is assuming that region is always present, but that's not true universally (I guess it will depend on country?)

LordShadowen, 2020-05-18:
Forgot to mention - the error reported in HA is "Failed to connect" and not "API changed", which I don't think is expected in this situation

jrester, 2020-05-19:
Thanks for reporting the error!

Yea, the library parses the json and constructs a response object to provide a common interface for accessing the response data.. But Tesla is sadly quite inconsistent with the response structure in different countries and across different versions.

What version is your powerwall currently running on? You can check either in the web interface or with `power_wall.get_version()`.

As for HA we already updated the error handling and reporting to handle such cases. The changes should go live in 1.10.0.

LordShadowen, 2020-05-19:
1.46.0

I did a quick and dirty test from the cloned repo, and it seems to work fine. 
I'll wait until all of it makes it into HA :)

jrester, 2020-06-05:
Update should be available as of 0.111 next week.

jrester, 2020-06-11:
@LordShadowen 0.111 should be available now, could you verify everything is working?

LordShadowen, 2020-06-12:
yes, it works great now, thanks
I noticed an oddity, not sure if it's worth opening an issue over:  all the sensors start with powerwall_ (which is great, I love that), except for binary_sensor.grid_status. Should it not be binary_sensor.powerwall_grid_status for consistency?

jrester, 2020-06-12:
Great!
You are right, that would be better for consistency and also signal that this sensor belongs to the powerwall integration. I will look into it!

closed 2020-06-12 by jrester