jrester/tesla_powerwall #10: Why are the methods for setting the mode commented out?

danjenkins, 2020-06-27:
I'd like to set the mode via the api using home assistant ... Wondering why the mode setting methods are commented out before I go try adding them back in myself

jrester, 2020-06-28:
I commented it out because it wasn't clear for me which values are required and if everything works as expected. Although it is documented in [vlaschiavo's](https://github.com/vloschiavo/powerwall2) documentation I am not sure how far it is still correct. 
If you would like to investigate this further I would be happy to merge a PR.

danjenkins, 2020-06-28:
Thanks! Will test the values using just a rest call... I've read mixed reports heather it works 100% of the time locally.... :(

Pyromancer579, 2020-07-05:
I too want to be able to set the mode via the API.

Please let me know once you have tested and established which values do what.

I just want to toggle between whatever mode it is in and 'Back-Up Only' as there are times when I don't want the Powerwall to provide energy (to stop my EV charger draining it)

jrester, 2020-12-10:
Closing this because of inactivity. Feel free to reopen if there are any updates!

closed 2020-12-10 by jrester