jrester/tesla_powerwall #14: Integration with TeslaMate?

natrlhy, 2020-09-04:
I will be getting Tesla Solar and Powerwall installed in a few months. I'd love to be able to collect and report on the statistics and usage outside of the Tesla App. I'm no programmer, etc but was wondering how this type of project could be incorporated into an awesome project like https://github.com/adriankumpf/teslamate ?

Thanks!

jrester, 2020-09-10:
The Problem is that thid API is written in python but Teslamate is written in elixir. But I don't think it would be to difficult to rewrite the API in elixir which could than be used by Teslamate.

jrester, 2020-09-22:
Closing this in favour of the [issue in the teslamate repo](https://github.com/adriankumpf/teslamate/issues/877)

closed 2020-09-22 by jrester