jrester/tesla_powerwall #26: API 0.3.10 and Mac OS X Big Sur

tmstiller, 2021-05-15:
Cannot login to Tesla Gateway running software version 21.13.2.
My credentials work with older versions os API on both Raspberry PI and Mac OS X Catalina but no combination of password and email address will login.
 Response to one attempt copied below.

/Users/tms/PycharmProjects/Test/login\ v-0.3.10.py 
Traceback (most recent call last):
  File "/Users/tms/PycharmProjects/Test/login v-0.3.10.py", line 41, in <module>
    powerwall.login_as(User.CUSTOMER, pw, ui)
  File "/usr/local/lib/python3.9/site-packages/tesla_powerwall/powerwall.py", line 64, in login_as
    response = self._api.login(user, email, password, force_sm_off)
  File "/usr/local/lib/python3.9/site-packages/tesla_powerwall/api.py", line 155, in login
    return self.post(
  File "/usr/local/lib/python3.9/site-packages/tesla_powerwall/api.py", line 145, in post
    return self._process_response(response)
  File "/usr/local/lib/python3.9/site-packages/tesla_powerwall/api.py", line 89, in _process_response
    self._handle_error(response)
  File "/usr/local/lib/python3.9/site-packages/tesla_powerwall/api.py", line 65, in _handle_error
    raise AccessDeniedError(
tesla_powerwall.error.AccessDeniedError: Access denied for resource /api/login/Basic: bad credentials: Login Error

jrester, 2021-05-15:
Tesla might have changed the login API again. I will take a look at it.

tmstiller, 2021-05-15:
Thanks, I tried all three forms of login with switching email and password.  No joy.

tmstiller, 2021-05-15:
Oops

tmstiller, 2021-05-15:
Dumb mistake

jrester, 2021-05-16:
Can you login in the web UI of the Powerwall?

tmstiller, 2021-05-16:
On May 16, 2021, at 5:04 AM, jrester ***@***.***> wrote:
> 
> Can you login in the web UI of the Powerwall?

Yes, as well as from a Mac running an older version of the API and a Raspberry ***@***.*** running a somewhat newer version.

I don't know if it's OS X 11.3.1 (Big Sur) or the new API. I'll boot up from an older version of OS X and install the new API to see it that works.

-- 
Tom Stiller ¯\_(ツ)_/¯

Facts are stubborn, but statistics are more pliable.
Mark Twain

tmstiller, 2021-05-16:
On May 15, 2021, at 1:31 PM, jrester ***@***.***> wrote:
> 
> Tesla might have changed the login API again. I will take a look at it.

Finally got USB backup drive booted into OS X 10.15.7 (Catalina), uninstalled API version -0.3.6 and installed version 0.3 10.

Everything works so it looks like OS X 11's (Big Sur) new security features may be the problem.

Thanks for taking time to respond to my problem

-- 
Tom Stiller ¯\\_(ツ)_/¯

Foreign aid might be defined as a transfer of money from poor people in
rich countries to rich people in poor countries.-- Douglas Casey, Classmate
of Bill Clinton at Georgetown University

jrester, 2021-05-16:
Ok, glad to hear it is working again.

tmstiller, 2021-05-16:
On May 16, 2021, at 11:14 AM, jrester ***@***.***> wrote:
> 
> Ok, glad to hear it is working again.

I've been trying to track down the problem with OX X 11.1 but I'm getting lost.
Using PyCharm I've constrained the problem between the first and second breaks at
***@***.***
On OS X 10.15 the 1st break shows a request POST and the 2nd a response of '200'

On OS X 11.1 the 1st break shows a request POST and the 2nd a response of '401'

I just cannot find my way through the 1st and 2nd breaks.

-- 
Tom Stiller  ( ͡° ͜ʖ ͡°)

A competent leader can get efficient service from poor troops, while on the contrary an incapable leader can demoralize the best of troops.

	John J. Pershing

jrester, 2021-05-17:
Could you try executing this command in a terminal. Replace `<ip>` and `<password>` with the correct values for your powerwall.

`$ curl -L -k -X POST -H "Content-Type: application/json" -d '{"password": "<password>", "username": "customer", "force_sm_off": false}' https://<ip>/api/login/Basic`

tmstiller, 2021-05-17:
On May 17, 2021, at 4:17 AM, jrester ***@***.***> wrote:
> 
> 
> Could you try executing this command in a terminal. Replace <ip> and <password> with the correct values for your powerwall.
> 
> $ curl -L -k -X POST -H "Content-Type: application/json" -d '{"password": "<password>", "username": "customer", "force_sm_off": false}' https://<ip>/api/login/Basic

That fails with 
{"code":401,"error":"bad credentials","message":"Login Error"}
on both OS X 10.15 and 11.1

-- 
Tom Stiller ¯\_(ツ)_/¯

Be thankful we're not getting all the government we're paying for.  -- Will Rogers

tmstiller, 2021-05-17:
On May 17, 2021, at 4:17 AM, jrester ***@***.***> wrote:
> 
> Could you try executing this command in a terminal. Replace <ip> and <password> with the correct values for your powerwall.
> 
> $ curl -L -k -X POST -H "Content-Type: application/json" -d '{"password": "<password>", "username": "customer", "force_sm_off": false}' https://<ip>/api/login/Basic

I screwed up.  I dropped a character from the password and copied the incorrect lint across systems.

The *correct* command produces:
{"email":"","firstname":"Tesla","lastname":"Energy","roles":["Home_Owner"],"token":"WCp5xmnkJYcwgUJzgQXXSDYeEE7LlV7GL5EiaIi8XL1ijDK3LLEryiipdidJwkcblzJQhpP-C1Wjm4P7VCvwJA==","provider":"Basic","loginTime":"2021-05-17T18:21:56.419529457-04:00"}%  

on OS X 10.15
and
{"email":"","firstname":"Tesla","lastname":"Energy","roles":["Home_Owner"],"token":"Hl_IdSFGxktoujC6aPQ9BvGlb8_h374TG9GjDEkxuqnNA6dGnDQhFWNa1SRitJ3MPCkZUEAmrxzU6ktsmrPnig==","provider":"Basic","loginTime":"2021-05-17T18:19:41.750851544-04:00"}%                                                                                                                    tms@MacBookPro ~ % 
 on OS X 11.1

I apologize for wasting your time.

-- 
Tom Stiller  ¯\_(ツ)_/¯

Politicians are like diapers; they need to be changed often and for the same reason.  -Mark Twain

jrester, 2021-05-18:
It's alright. Happens to all of us.

I cannot identify any issue here. Can you provide me the source code of your script and send me the output of executing the `get_status` method?

tmstiller, 2021-05-18:
> On May 18, 2021, at 12:12 PM, jrester ***@***.***> wrote:
> 
> It's alright. Happens to all of us.

Thanks

> I cannot identify any issue here. Can you provide me the source code of your script and send me the output of executing the get_status method?

It's all good. The faulty password was copied and pasted into several tests without me ever noticing the glitch. Once it was cleaned up everything functioned as expected.

Once again, I apologize fir the dust-uo.

-- 
Tom Stiller ¯\_(ツ)_/¯

Superior people are only those who let it be discovered by others; the need to make it evident forfeits the very virtue they aspire to. - Sidney J. Harris

jrester, 2021-05-18:
Ok, than I am closing this issue.

closed 2021-05-18 by jrester