jrester/tesla_powerwall #27: Changes to new Version 21.20.6

tq23ld, 2021-08-12:
Yesterday my powerwall 2 updated to V 21.20.6. 

Prior to that I could get all the documented GETS without authentication with just power_wall = Powerwall("192.168.0.15") - and have been using that in  a script for extracting powerwall info into a dbase.

Yesterdays version update required me to go through the procedure to power off/on my powerwall and enter a new password. This worked and gives access to the GUI when putting my powerwall local IP address into a browser. 

However adding  power_wall.login("<myemail>", "<password>") to my script and using the same email and pw which works when connecting to the GUI produces  "Access denied for resource /api/login/Basic: bad credentials: Login Error"

I have tried a number of alternatives such as power_wall.login_as(User.CUSTOMER, "<myemail>", "<password>")  but still gives Access denied error
N.B. 'customer' in lower case gives 
File "/usr/lib/python3.7/enum.py", line 349, in __getattr__
    raise AttributeError(name) from None

jrester, 2021-08-13:
Have you made sure you supplied your credentials in the right order? The first argument is the password and the second your email (which can be ommitted) like so:
```python
powerwall.login("<password>", "<email>")
```

tq23ld, 2021-08-13:
No I was not supplying my credentials in the right order and now it works perfectly - thank you! The reason for my apparent stupidity is that I originally got all my info from  https://opensourcelibs.com/lib/tesla_powerwall where the order (and a few other things) are incorrectly documented and I didn't think to check it with this site.  Thanks again.

closed 2021-08-13 by tq23ld