jrester/tesla_powerwall #28: Slightly wrong Dokumentation

kl1947, 2021-08-17:
First: Thank you, jrester, for this module!

In the doku part "Authentication":
powerwall.login("<password>", "<email>")
powerwall.login_as(User.INSTALLER, "<password>", "<email>")
should read:
powerwall.login("<email>","<password>" )
powerwall.login_as(User.INSTALLER, "<email>", "<password>") 

Another hint, NO ISSUE:
powerwall = Powerwall("<ip of your Powerwall>")
Powerwall Firmware 20.49.x worked with a locally resolvable NAME as ip, Firmware 21.20.6 requires the ip address. So the Doku is correct.

jrester, 2021-08-17:
Hey there,

I am sorry but I can't see any difference between the two.
Thanks for reporting the part about the ip and hostname. I will add it to the doumentation!

jrester, 2021-09-17:
Closing this because of inactivity

closed 2021-09-17 by jrester