jrester/tesla_powerwall #38: Can't login, bad credentials.

kemble9900, 2022-04-30:
Hi, just started with this as I wanted to start on a project with this library, it looks good but I can't get it to login.

I'm currently using this bit of code to test logging in
`from tesla_powerwall import Powerwall

powerwall = Powerwall("192.168.0.164")

powerwall.login("password", "email")
powerwall.is_authenticated()

powerwall.get_charge()`

and im getting the error
`Traceback (most recent call last):
  File "C:/Users/caleb/OneDrive/Python/tesla.py", line 5, in <module>
    powerwall.login("password", "email")
  File "C:\Users\caleb\AppData\Local\Programs\Python\Python39\lib\site-packages\tesla_powerwall\powerwall.py", line 66, in login
    return self.login_as(User.CUSTOMER, password, email, force_sm_off)
  File "C:\Users\caleb\AppData\Local\Programs\Python\Python39\lib\site-packages\tesla_powerwall\powerwall.py", line 59, in login_as
    response = self._api.login(user, email, password, force_sm_off)
  File "C:\Users\caleb\AppData\Local\Programs\Python\Python39\lib\site-packages\tesla_powerwall\api.py", line 154, in login
    return self.post(
  File "C:\Users\caleb\AppData\Local\Programs\Python\Python39\lib\site-packages\tesla_powerwall\api.py", line 144, in post
    return self._process_response(response)
  File "C:\Users\caleb\AppData\Local\Programs\Python\Python39\lib\site-packages\tesla_powerwall\api.py", line 88, in _process_response
    self._handle_error(response)
  File "C:\Users\caleb\AppData\Local\Programs\Python\Python39\lib\site-packages\tesla_powerwall\api.py", line 64, in _handle_error
    raise AccessDeniedError(
tesla_powerwall.error.AccessDeniedError: Access denied for resource /api/login/Basic: bad credentials: Login Error`

I've checked the username and password details are correct, which they are, I haven't noticed any errors in what I've written. I am aware that there is also a self.login_as(User.CUSTOMER, password, email) login method, however I wasn't sure what I should be inputting in the place of User.CUSTOMER

If anyone has any idea what I'm doing wrong that would be greatly appreciated.
Many thanks, kemble9900.

jrester, 2022-04-30:
Could you please provide the version of the powerwall libary you are using: `pip3 show tesla_powerwall` 
As well as the version of your Powerwall: open `https://192.168.0.164/api/status` in a browser

kemble9900, 2022-04-30:
The library version is 0.3.17 and the powerwall is version 22.9.1 bce98cad

jrester, 2022-04-30:
Interessting, I have the same version combo and it is working for me. Could you try if it is working from the terminal, to identify wheter it is a problem with the libary: https://github.com/vloschiavo/powerwall2#post-apiloginbasic

kemble9900, 2022-04-30:
I'm not sure if I was doing something wrong or if it's not working but I put

`C:\Users\user>curl --cacert cacert.pem -s -i -X POST -H "Content-Type: application/json" -d '{"username":"","password":"PasswordSerial","force_sm_off":false}' https://192.169.0.164/api/login/Basic
`

into command prompt but didn't get any output, it just processed for a few seconds and gave no output.

jrester, 2022-04-30:
No output is very weird. Could you try replacing the `--cacert cacert.pem` with `-v -k`?

kemble9900, 2022-04-30:
That did give an output, I got

`* Could not resolve host: k
* Closing connection 0
*   Trying 192.169.0.164:443...
* connect to 192.169.0.164 port 443 failed: Timed out
* Failed to connect to 192.169.0.164 port 443 after 21034 ms: Timed out
* Closing connection 1`

not sure what the issue could be about the port

jrester, 2022-04-30:
`Could not resolve host: k` normally means you forgot the `-` in front of the `k`
The port issue is probably because you are trying to access 192.**169**.0.164 but your powerwall has the IP 192.**168**.0.164

kemble9900, 2022-04-30:
sorry about that, must have typed it wrong, trying again I got

`C:\Users\user>curl -v -k -s -i -X POST -H "Content-Type: application/json" -d '{"username":"","password":"PasswordSerial","force_sm_off":false}' https://192.168.0.164/api/login/Basic
*   Trying 192.168.0.164:443...
* Connected to 192.168.0.164 (192.168.0.164) port 443 (#0)
* schannel: disabled automatic use of client certificate
* schannel: using IP address, SNI is not supported by OS.
* schannel: ALPN, offering http/1.1
* schannel: ALPN, server accepted to use http/1.1
> POST /api/login/Basic HTTP/1.1
> Host: 192.168.0.164
> User-Agent: curl/7.79.1
> Accept: */*
> Content-Type: application/json
> Content-Length: 64
>
* Mark bundle as not supporting multiuse
< HTTP/1.1 401 Unauthorized
HTTP/1.1 401 Unauthorized
< Cache-Control: no-cache, no-store
Cache-Control: no-cache, no-store
< Content-Type: application/json
Content-Type: application/json
< X-Content-Type-Options: nosniff
X-Content-Type-Options: nosniff
< Date: Sat, 30 Apr 2022 13:54:48 GMT
Date: Sat, 30 Apr 2022 13:54:48 GMT
< Content-Length: 62
Content-Length: 62

<
{"code":401,"error":"bad credentials","message":"Login Error"}* Connection #0 to host 192.168.0.164 left intact`

jrester, 2022-04-30:
If even curl fails this means that either your credentials are incorrect or Tesla changed the login process.

jrester, 2022-05-06:
@kemble9900 any updates on this?

closed 2022-05-21 by kemble9900