jrester/tesla_powerwall #39: Charge and Capacity reflect only 1 battery in array of batteries.

T3CHKOMMIE, 2022-05-23:
Powerwall.get_energy and get_charge seem to be making assumptions when more than 1 powerwall is installed.

I ittereated through batteries and calculated total charge and capacity now which i think is more accurate. Not sure if this is a powerwall issue or something with the library. 

Powerwall Charge: 36%
Get_Charge 5075
Get_Capacity 14085
Battery 1: charge: 2358 capacity: 14150  SN: TG121***********
Battery 2: charge: 5073 capacity: 14085 SN: TG121***********
Calculated Capacity: 28235
Calculated Energy Remaining: 7431
Calculated %: 26.3

its almost a 10% difference between what is reported and what is calculated. 

when two powewalls are in parallel, the capacitance is added and voltage remains constant. it seems like Get_Charge and Get_Capacity take the last element in the list? 

I know it doesnt help my powerwalls are unbalanced >_< figured id file something here before I bothered Tesla (FWIW my mobile app also seems to be reporting wrong numbers... probably something with API or PW)

jrester, 2022-05-23:
The functions `get_charge` and `get_capacity` just read the values from API response, so the Powerwall is indeed sending wrong values. It looks like tesla is calculating the values wrong. 
I believe the best option would be to update the functions and, as you did, calculate the values based on the single values of each battery.
Thank your very much for reporting!

T3CHKOMMIE, 2022-05-23:
Great, I just got off the call with Tesla. They are looking into it. Looks like one of  my powerwalls doesnt want to charge or discharge at the same rate as the other. 

great library btw. Ill try to contribute when/where i can :)

T3CHKOMMIE, 2022-05-23:
Eureka, it looks like the API is reporting my capacity and energy as a sum of the batts now:

Powerwall Charge: 18%
5130
28236
Battery 1: 2350 14151 - TG1213**********
Battery 2: 2780 14085 - TG1213**********
Calculated Capacity: 28236
Calculated Energy Remaining: 5130
Calculated %: 18.2

Tesla said I had a battery that was performing a firmware update. Might just have to chalk this up to timing :)

jrester, 2022-05-24:
Alright, then I am going to close this. But nevertheless interessting that this can happen :)