jrester/tesla_powerwall #47: Error: tesla_powerwall.error.PowerwallUnreachableError

pietervanderwesthuizen, 2023-06-06:
I get Powerwall is unreachable error with below code:

from tesla_powerwall import Powerwall, User, API

password='xxx'
email='xxx'
ip = "xxx" # Address of your Powerwall Gateway

powerwall = Powerwall(ip)

powerwall.login(password, email)

Console:
> reticulate::source_python('C:/Users/piete/OneDrive/Documents/R/Powerwall 2.py')
Error: tesla_powerwall.error.PowerwallUnreachableError: Powerwall is unreachable: HTTPSConnectionPool(host='192.168.1.109', port=443): Max retries exceeded with url: /api/login/Basic (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1002)')))
>

pietervanderwesthuizen, 2023-06-11:
> Hi, this error seems to come from the self-signed certificate provided by the powerwall. Although this powerwall lib skips the certificate verification, it does not seem to work in your case. For further investigation could you please test the following short code snippet and report if it also throws the same error or returns a <Response [200]>:
>
> import requests
> print(requests.get("https:// <https://%3cinsert%20your%20powerwall%20ip%3e> <insert your powerwall ip>", verify=False))

<Response [200]>

ezorzin, 2023-06-16:
Hi, on my system I have exactly the same behavior: <Response [200]>.
Any news?

jrester, 2023-08-05:
@pietervanderwesthuizen and  @ezorzin, sorry for my late reply. Does the error still occur?

jrester, 2023-10-19:
Closing this because of inactivity

closed 2023-10-19 by jrester