jrester/tesla_powerwall #48: Feature Request: Per Phase Meter Data

dh99999, 2023-07-19:
Hi,

Just wondering if it would be possible to add additional detail to the meter statistics?

The values I am particularly interested in are the per-phase power and voltage values for the site and solar meters from the https://X.X.X.X/api/meters/site and https://X.X.X.X/api/meters/solar APIs respectively:

- real_power_a
- real_power_b
- real_power_c
- v_l1n
- v_l2n
- v_l3n
  
Sample from site meter:

>	{
		"id": 0,
		"location": "site",
		"type": "synchrometerX",
		"cts": [
			true,
			true,
			true,
			false
		],
		"inverted": [
			false,
			false,
			false,
			false
		],
		"connection": {
			"short_id": "XXXX",
			"device_serial": "YYYYY",
			"https_conf": {}
		},
		"Cached_readings": {
			"last_communication_time": "2023-07-17T16:27:36.390511647+10:00",
			"instant_power": -22,
			"instant_reactive_power": -1154,
			"instant_apparent_power": 1154.2096863222037,
			"frequency": 0,
			"energy_exported": 2502052.53320083,
			"energy_imported": 2970955.180779687,
			"instant_average_voltage": 409.3644348662122,
			"instant_average_current": 5.529,
			"i_a_current": 3.02,
			"i_b_current": 1.311,
			"i_c_current": 1.198,
			"last_phase_voltage_communication_time": "2023-07-17T16:27:36.390568769+10:00",
			"v_l1n": 235.82,
			"v_l2n": 234.34,
			"v_l3n": 238.88,
			"last_phase_power_communication_time": "2023-07-17T16:27:36.390511647+10:00",
			"real_power_a": -144,
			"real_power_b": 48,
			"real_power_c": 74,
			"reactive_power_a": -583,
			"reactive_power_b": -297,
			"reactive_power_c": -274,
			"last_phase_energy_communication_time": "0001-01-01T00:00:00Z",
			"serial_number": "JBL19236C4D232",
			"timeout": 60000000000,
			"instant_total_current": 5.529
		}
	}
]

Happy to test/assist if I can.

jrester, 2023-08-05:
Hey @dh99999, I think the endpoints would be a great addition! I will take a look at the API endpoints and let you know, as soon as it is implemented.

dh99999, 2023-08-06:
That would be great - thanks!

Happy to help test!

jrester, 2023-08-17:
Hi @dh99999, it is now implemented in #51. Is it possible that you test it if everything works as expected?

dh99999, 2023-08-18:
Hi @jrester,

Looks pretty good to me - a couple of comments:

1) Would it be possible to have MeterDetails() inherit from Meter() so that a client can just call get_meter_site() in order get both the aggregated data and the detail data in the one call to the powerwall?

The response to the meters/site call contains the aggregate data as well.

2) Consider adding MeterDetails to tesla_powerall/__init__.py so it can be used externally.

3) (Just an FYI) The response from my powerwall for meters/solar looks to have a bug - it will always leave out one of real_power_a, real_power_b or real_power_c. 

It is not consistent, but makes testing a pain!

dh99999, 2023-08-18:
Not 100% sure about this, but their also might be an issue in powerwall.py

`    def get_meter_site(self):
        meter_response = self._api.get_meters_site()
        if meter_response is None or len(meter_response) == 0:
            raise ApiError("The powerwall returned no values for the site meter")`

Pylance is reporting that ApiError is not defined.

jrester, 2023-08-21:
Thanks for your great input! I will continue to work on it and keep you updated!

jrester, 2023-09-16:
@dh99999 After a bit of delay I finally finished this feature and it is now available in version v0.4.0

dh99999, 2023-09-18:
Looks good to me! 

Thanks!

closed 2023-10-18 by jrester