jrester/tesla_powerwall #49: Site and Solar Votages being misreported

ben-j-s, 2023-07-23:
I have just set up a script that uses your API and I am seeing some strange data being reported from the Powerwall. We have a single phase install in Australia so our voltage should be around 240V. The site and solar voltages are being reported at around 420V (which is what a 3-phase supply should be). The current seems to be correct for the site (if you multiply the current by the expected voltage, not the reported voltage you get the observed wattage) but the current for the battery (which is showing the correct AC voltage) seems to be reporting what I'm guessing is the DC current, not the AC current (it works out to 48V if you divide the wattage by the current).

My Gateway is running 23.12.2 (63bc46d8) and we have a single battery and solar on the backup circuit. I have attached an output of get_meters(), please let me know if you need anything else.

https://github.com/jrester/tesla_powerwall/files/12139105/get_meters.txt

jrester, 2023-08-05:
Hey @ben-j-s , sorry for my late response and thanks for your report.
If the values from `get_meters` are wrong, that basically means that the powerwall is reporting the wrong values. No  post processing is done on the values reported by the powerwall (except if you use e.g. `Meter.get_instant_total_current` the result will be rounded, but this would be unrelated to your issue).
If the values are always reported wrong by the powerwall and there is a clear way to calculate the correct values, I could implement something to automatically transform them.

closed 2023-12-20 by jrester

falcon1988, 2023-12-22:
I've just recently had a Powerwall 2 with a Backup Gateway 2 installed in Australia and I'm seeing the same behaviour as @ben-j-s.

Interestingly the new calls added for #48 (get_meter_site() and get_meter_solar()) report both the incorrect voltage value on variable 'instant_average_voltage' and the correct voltage value on 'v_l1n'.

My guess is that my Gateway incorrectly thinks that it is monitoring three phases instead of just one and is multiplying v_l1n by the square root of 3. Other values reported by the API such as 'instant_power' and 'instant_average_current' appear to be consistant with 'v_l1n' being the correct voltage (i.e. multiplying 'v_l1n' by 'instant_average_current' gives the same value as is reported by 'instant_power')

This definitely appears to be either a misconfiguration (I'll be following up with my installer) or a bug in Tesla's firmware, so I'm not sure if you would want to correct for this somehow in your code, but I'm adding this comment for anyone else who might run into the same issue.