jrester/tesla_powerwall #52: Feature request: add ability to set backup percentage

greyghoster, 2023-08-18:
I have been dealing with to peak usage periods, morning and night and want to programmatically control the backup percentage to _release_ more power during the peak period. Can that function be added?

jrester, 2023-08-21:
Hey @greyghoster, I would be happy to add such a feature. This was already discussed in #20, but the result of the investigation was that you need to switch to installer mode, which to my knowledge requires physically activating a switch and therefore does not make sense for any automation. If this is no longer required/there is a simple solution I would be happy to implement this feature.

jrester, 2023-09-16:
Closing this because of inactivity

closed 2023-09-16 by jrester