jrester/tesla_powerwall #54: Enabling/disabling grid charging

DanielBaulig, 2023-10-10:
Hi, I am looking to add a feature to the Tesla Powerwall Home Assistant integration and I believe this is the API being used in there.
I would like to be able to toggle grid charging on and off, like is possible within the Tesla app. I'm not sure if this is something that Powerwall exposes locally or not, but thought I'd ask here before going digging around too much.

jrester, 2023-10-15:
Hey @DanielBaulig, I am not sure but could this be the feature you are looking for: https://github.com/jrester/tesla_powerwall#off-grid-status-set-island-mode?

DanielBaulig, 2023-10-15:
@jrester I don't believe that is what I am looking for. If I understand the island mode feature correctly, then this will disconnect the system from the grid and power the home from the battery -- even when the grid isn't down.

However, what I am looking for is the "Grid charging" setting (or equivalent) from under "Settings -> Powerwall -> Grid charging" in the Tesla app. It allows the Powerwall use the grid for charging instead of just solar. This can be enabled or disabled but I believe may also be subject to installer configuration (i.e. some people might not be able to directly control this setting due to regulatory limitations).

Hope this clarifies it. If I am misunderstood please let me know!

jrester, 2023-10-18:
@DanielBaulig Ah yeah, the island mode feature is something different. Sadly, I could not find any endpoint in the [local API documentation](https://vloschiavo.github.io/powerwall2/openapi/#/) that looks like it could control the Grid Charging setting. Tesla really locked down the options for local control of their powerwalls and it seems like most of their features are only available through their public Tesla Motor Owners API.

Nevertheless, there is an option in the [teslajsonpy library](https://github.com/zabuldon/teslajsonpy/blob/dev/teslajsonpy/energy.py#L251) (which uses the public Tesla Motor Owners API) that can enable grid charging. This library is also used in [the custom tesla homeassistant ](https://github.com/alandtse/tesla) integration. Therefore, I believe the best option is to add this feature to the mentioned custom integration.

jrester, 2023-11-14:
Closing this due to inactivity

closed 2023-11-14 by jrester