jrester/tesla_powerwall #57: FEATURE REQUEST - Separate data for multiple PW batteries connected to a single gateway

ejbatts, 2024-01-06:
Is it possible to extract data from separate PW batteries attached to a single gateway? I have 2 PW batteries connected to separate phases of a 3 phase installation. Specifically, I would like to know Battery AC Power (charge/discharge rate) of each battery so I can deduce total load on each phase.

This separate data is available on //[gateway ip address]/system page, as well as voltage and state of charge, but I don't know how to extract it in HA. The system page is only accessible if logged in as the installer, and not as the customer. Logging in as an installer requires a physical toggle of a battery on/off switch

The individual data points visible on the installer system page are:
- Powerwall State
- Charge Level
- AC Voltage
- AC Power

bubonicbob, 2024-01-06:
Thanks for creating this issue.    It would be really helpful if you would get [this script](https://github.com/vloschiavo/powerwall2/blob/master/samples/powerwallstats.sh) working locally and start fetching data from your powerwall to call out specifically which URL and which data you are after.   This is a shell script that should work in your mingw terminal as long as you make sure you have curl installed.

[The directory](https://github.com/vloschiavo/powerwall2/tree/master/samples) where that script is has all sorts of json files that are probably named for the exact query you can feed into that script.

ejbatts, 2024-01-06:
I don't think this will work in the current format. The issue is that the data I require seems to only be available after installer login which requires a physical switch toggle I will keep hunting though.

bubonicbob, 2024-01-06:
[This script](https://github.com/vloschiavo/powerwall2/blob/master/scripts/getsamples.sh) looks like it is pretty neat too since it does a complete dump of all apis... even the ones that require special temporary modes.

ejbatts, 2024-01-08:
I am going around in circles trying to run these scripts. First major hurdle seems that I cant install JQ which is required. Also finding that inbuilt commands such as 'from' are not found. Running git bash app on Windows, is that the problem?

Have also discovered that the data I am seeking to expose is also seen on the Tesla Pros App which does not require a physical toggle, but does require connecting to the Gateway wifi (TEG-xxx)

bubonicbob, 2024-01-08:
I would stick to [the first](https://github.com/vloschiavo/powerwall2/blob/master/samples/powerwallstats.sh) script.   The other script seems to reboot your gateway and, while *probably* safe, I can't vouch for it.

Windows is sure a tough platform to use when working with open source projects such as this.    Perhaps you'd like something like [chocolatey](https://community.chocolatey.org/packages/jq) to help with installing these utilities?   I've never used it before so YMMV

bubonicbob, 2024-01-08:
> I am going around in circles trying to run these scripts. First major hurdle seems that I cant install JQ which is required. Also finding that inbuilt commands such as 'from' are not found. Running [...]

The idea of working with these scripts is that, if they can't fetch the data you want, neither can we in this library.  Special buttons and special connections directly to the gateway's WiFi are going to make it impractical if not possible to support without it being so unreliable that it isn't worth investigating.

ejbatts, 2024-01-10:
@jrester I am following the instructions on #20 comment but when I download and try to modify the request.bin file i am seeing this

9
1232100-10-E--CN321329G1E123z

config.json

so not able to 'replace the din'.

Also, where do I store the request.bin file, in my user/<username> folder?

zigam, 2024-01-10:
@ejbatts the data you need should be available from `//[gateway ip address]/api/system_status` and shouldn't require installer login.  Give that a try.

ejbatts, 2024-01-10:
The command `//192.168.1.41/api/system_status` returns:
1. in Git Bash bash: **//192.168.1.41/api/system_status: No such file or directory**
2. in Python: **SyntaxError: invalid syntax**
3. In PowerShell: 
//192.168.1.41/api/system_status : The term '//192.168.1.41/api/system_status' is not recognized as the name of a
cmdlet, function, script file, or operable program. Check the spelling of the name, or if a path was included, verify
that the path is correct and try again.
At line:1 char:1
+ //192.168.1.41/api/system_status
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (//192.168.1.41/api/system_status:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

I am VERY new to this, sorry!

zigam, 2024-01-10:
@ejbatts I meant access that as a web page (so in your case https://192.168.1.41/api/system_status), after logging in to the gateway as a customer.  If that works, you can figure out a programmatic way to access it.  I'm not sure how HA integration works, but that page will give you json output that should have all the required information.

bubonicbob, 2024-01-10:
@ejbatts I really feel like you should invest in getting a basic Linux VM running in docker.  or maybe borrow a Mac.  it would really make things easier.

JoshingtonState, 2024-01-11:
Another open for you guys is to use your homeassistant Terminal and SSH add on.   You can then use the commands in the shell:

```
wget https://raw.githubusercontent.com/vloschiavo/powerwall2/master/samples/powerwallstats.sh
chmod u+x powerwallstats.sh
export POWERWALLIP=yourip
export PASSWORD=yourpass
./powerwallstats.sh /api/meters/aggregates
```

ejbatts, 2024-01-11:
@zigam @bubonicbob @JoshingtonState
I really appreciate everyone's suggestions to this newbie. 

Been copping some flak from SWMBO about the time I am spending on this, so will get back to it soon. lol.

@bubonicbob I have already installed Ubuntu on my PC using WSL. I guess you are telling me to use the Ubuntu shell for all the above stuff? My other option is to install it on a VM on my NAS providing I have enough spare memory to run it side by side with my Home Assistant VM and my Plex Server which run 24/7. I tried the Ubuntu shell at some stage but struggled to find the correct path to installed modules?

Thanks again

ejbatts, 2024-01-11:
> @ejbatts I meant access that as a web page (so in your case https://192.168.1.41/api/system_status), after logging in to the gateway as a customer. If that works, you can figure out a programmatic [...]

@zigam I tried this previously but still could not find the separated battery data that I am after, only the aggregated data. I will try again however

ejbatts, 2024-01-11:
> Another option for you guys is to install Homeassistant “Terminal & SSH” add on which gives you a terminal option in your main menu.

@JoshingtonState I have these installed on HA already and will tryout you suggestions, thanks

JoshingtonState, 2024-01-11:
> @zigam @bubonicbob @JoshingtonState I really appreciate everyone's suggestions to this newbie.
>
> Been copping some flak from SWMBO about the time I am spending on this, so will get back to it soon. [...]

Yeah I would absolutely just use a shell under Ubuntu.   Seems  like Ubuntu uses apt for package management, so you can do things like:

```
sudo apt install wget curl jq nano
```

Then all the commands I mentioned above work as-is.

zigam, 2024-01-11:
> @zigam I tried this previously but still could not find the separated battery data that I am after, only the aggregated data. I will try again however

Maybe post the output here and I can help analyze.

ejbatts, 2024-01-11:
> @ejbatts Maybe post the output here and I can help analyze.

The data is here! I missed it first time round. Have added some line breaks to help see it more clearly. The data I _need_ to poll frequently in HA is "p_out" for both batteries. "nominal_energy_remaining" and "nominal_full_pack_energy" would also be handy for each battery.

Only a customer login is required.

> {"command_source":"Configuration","battery_target_power":524.5,"battery_target_reactive_power":0,"nominal_full_pack_energy":29859,"nominal_energy_remaining":21242,"max_power_energy_remaining":0,"max_power_energy_to_be_charged":0,"max_charge_power":10000,"max_discharge_power":10000,"max_apparent_power":10000,"instantaneous_max_discharge_power":17820,"instantaneous_max_charge_power":14000,"instantaneous_max_apparent_power":11000,"hardware_capability_charge_power":0,"hardware_capability_discharge_power":0,"grid_services_power":-0,"system_island_state":"SystemGridConnected","available_blocks":2,"available_charger_blocks":0,
> 
> "battery_blocks":
> 
> [{"Type":"","PackagePartNumber":"3012170-05-C","PackageSerialNumber":"TGxxxxxxxxxxxx","disabled_reasons":[],
> 
> "pinv_state":"PINV_GridFollowing","pinv_grid_state":"Grid_Compliant","nominal_energy_remaining":10498,"nominal_full_pack_energy":14724,**"p_out"**:260,"q_out":-1080,"v_out":245.70000000000002,"f_out":49.953,"i_out":-7.4,"energy_charged":2667511,"energy_discharged":2340336,"off_grid":false,"vf_mode":false,"wobble_detected":false,"charge_power_clamped":false,"backup_ready":true,"OpSeqState":"Active","version":"aa269d352a70b0"},
> 
> {"Type":"","PackagePartNumber":"3012170-05-E","PackageSerialNumber":"TGxxxxxxxxxxxx","disabled_reasons":[],
> 
> "pinv_state":"PINV_GridFollowing","pinv_grid_state":"Grid_Compliant","nominal_energy_remaining":10744,"nominal_full_pack_energy":15135,**"p_out"**:270,"q_out":-1010,"v_out":245.3,"f_out":49.953,"i_out":-7.6000000000000005,"energy_charged":583676,"energy_discharged":491331,"off_grid":false,"vf_mode":false,"wobble_detected":false,"charge_power_clamped":false,"backup_ready":true,"OpSeqState":"Active","version":"aa269d352a70b0"}],
> 
> 
> "ffr_power_availability_high":11000,"ffr_power_availability_low":11000,"load_charge_constraint":0,"max_sustained_ramp_rate":2500000,"grid_faults":[],"can_reboot":"Yes","smart_inv_delta_p":0,"smart_inv_delta_q":-2060,"last_toggle_timestamp":"2024-01-08T12:38:47.779647+11:00","solar_real_power_limit":1.7976931348623157e+308,"score":10000,"blocks_controlled":2,"primary":true,"auxiliary_load":0,"all_enable_lines_high":true,"inverter_nominal_usable_power":11000,"expected_energy_remaining":0}

bubonicbob, 2024-01-11:
> The data is here! I missed it first time round. Have added some line breaks to help see it more clearly. The data I _need_ to poll frequently in HA is "p_out" for both batteries. [...]

thanks for posting this!   one of the side effects of using a known script is that you can post your exact command line and full results.   the outcome is that the results are so actionable that it makes it easy to code up the changes and write tests against your exact results.  as of now, I couldn't code up much based on these results without personally reproducing your same scenario...  but as you're finding out, you have a pretty unique situation so the odds of you getting your system supported go down.

ejbatts, 2024-01-11:
My understanding is that the Tesla integration polls the Gateway using the local API, and currently extracts 32 entities, one being a sensor named "Battery Power". This is the aggregate, and I am asking for separate data from each battery being "p_out". I don't think my situation is unique in that hundreds/thousands of users have multiple batteries connected to a single gateway, and I am sure I am not alone in having them on different phases. I agree it would be less common that a user like me wants to use the data to deduce loads on individual phases.

Having said that, a user on another [thread](https://github.com/alandtse/tesla/issues/681) detected a battery failure only after logging in as an installer. he claims that having separated information constantly available would have allowed earlier detection. His battery had been in error state for over a year!

ejbatts, 2024-01-12:
It appears that [at least some of] the data supplied by the integration comes from the API branch "**api/meters/aggregates**"? The data I want appears in a different branch "**api/system_status**" and the individual battery data points are in a sub-sets of that branch headed by **"Type":"","PackagePartNumber":"xxxxxxx-xx-x","PackageSerialNumber":"TGxxxxxxxxxxxx"**
If that is correct, I am imagining it is possible to access that data. I suppose the data points with the same name from different sub-sets would need to be appended with a prefix or suffix, related to the serial number perhaps, to distinguish them.

I have no idea how to code this, but think it is possible?

jrester, 2024-01-12:
@ejbatts great to hear, that the data is also available over the "normal" API! This library does currently not use the `system_status` endpoint, but I see no problem in supporting it. I will take a look at it and see if it can be implemented easily and get back to you once it's done.

ejbatts, 2024-01-12:
@jrester In an effort to explore further, I am trying to install and run tesla_powerwall (on Ubuntu) as per the setup instructions but stumble here:

```
$ from tesla_powerwall import Powerwall
from: too many arguments
```

Chat GPT confirms the correct syntax.

bubonicbob, 2024-01-12:
> @jrester In an effort to explore further, I am trying to install and run tesla_powerwall (on Ubuntu) as per the setup instructions but stumble here:
> 
> ```
> $ from tesla_powerwall import Powerwall
> from: too many arguments
> ```
> 
> Chat GPT confirms the correct syntax.

Are you trying to run the example python code?   That won't work by just copy/pasting to the command line.

You've got to clone the rep and then follow the commands at the top of the instructions:
```
$ export POWERWALL_IP=<ip of your Powerwall>
$ export POWERWALL_PASSWORD=<your password>
$ python3 examples/example.py
```

Then you'll run whatever code is in example.py

ejbatts, 2024-01-12:
I installed the repo

`pip install tesla_powerwall`

is that not enough?

bubonicbob, 2024-01-12:
> I installed the repo
> 
> `pip install tesla_powerwall`
> 
> is that not enough?

Depends what you're trying to accomplish, I guess.   Those instructions written for somebody who wants to use the publicly available tesla_powerwall library to write some python code that uses it.    In your case, I assume you're looking to find extra fields that the library does not yet support, so running you own python code against the library won't really accomplish that.

Running the pip command installs the library so that if you write and run your own code locally then the library will be automatically located.

Perhaps it depends on who you're talking to, but I don't think you can do better than just running [this shell script](https://github.com/vloschiavo/powerwall2/blob/master/samples/powerwallstats.sh) that uses curl to authenticate, store the authentication cookie in a file, run whatever query you want against your powerwall, then spit out the raw results for you.   It is perfect for your Ubuntu environment.

bubonicbob, 2024-01-12:
Try these commands.   Replace `yourip` and `yourpass` with your info.

```
curl -O https://raw.githubusercontent.com/vloschiavo/powerwall2/master/samples/powerwallstats.sh
sed -i '' -e 's/POWERWALLIP=.*/POWERWALLIP="yourip"/g' powerwallstats.sh
sed -i '' -e 's/POWERWALL=.*/POWERWALLIP="yourpass"/g' powerwallstats.sh
chmod u+x powerwallstats.sh
```

Now you can fetch any path from your powerwall like so:

```
./powerwallstats.sh /api/powerwalls
```

ejbatts, 2024-01-12:
I am truly grateful for your help but feel I am way over my depth...

batts@STUDY:~$ curl -O https://raw.githubusercontent.com/vloschiavo/powerwall2/master/samples/powerwallstats.sh
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  3269  100  3269    0     0   9801      0 --:--:-- --:--:-- --:--:--  9787
batts@STUDY:~$ sed -i '' -e 's/POWERWALLIP=.*/POWERWALLIP="192.168.1.41"/g' powerwallstats.sh
**sed: can't read : No such file or directory**
batts@STUDY:~$ sed -i '' -e 's/PASSWORD=.*/PASSWORD="YES I HAVE MY PASSWORD HERE"/g' powerwallstats.sh
**sed: can't read : No such file or directory**

bubonicbob, 2024-01-12:
> I am truly grateful for your help but feel I am way over my depth...
>
> batts@STUDY:~$ curl -O https://raw.githubusercontent.com/vloschiavo/powerwall2/master/samples/powerwallstats.sh % Total % [...]

Sorry, I may have gotten too fancy with those commands.

Try these:
```
curl -O https://raw.githubusercontent.com/vloschiavo/powerwall2/master/samples/powerwallstats.sh
chmod u+x powerwallstats.sh
```

Use your favorite editor to edit the downloaded file named `powerwallstats.sh` to replace these lines with your information.    Make sure to keep the quotes.
```
POWERWALLIP="powerwall"			# This is your Powerwall IP or DNS Name
PASSWORD='Y0ur$up3r$3cr3tPassword!'	# Login to the Powerwall UI and Set this password - follow the on-screen instructions
```

Now you can run:
```
./powerwallstats.sh /api/powerwalls
```

bubonicbob, 2024-01-12:
> I am truly grateful for your help but feel I am way over my depth...
>
> batts@STUDY:~$ curl -O https://raw.githubusercontent.com/vloschiavo/powerwall2/master/samples/powerwallstats.sh % Total % [...]

If you want, I could make a zoom meeting with you in within the next hour

ejbatts, 2024-01-13:
ah sorry, been away from PC mowing lawns
I could do a zoom meeting but could also just wait until @jrester has a look at the code as per above...to get the new entities into HA integration.
I thought I could do a bit of exploring withpython/linux/git etc but its way more complicated/finicky than I thought. Up to you.

closed 2024-01-15 by jrester

jrester, 2024-01-15:
The feature is included in the [v0.5.1](https://github.com/jrester/tesla_powerwall/releases/tag/v0.5.1) release.
The feature can now be added to home-assistant. @bubonicbob Do you want to take care of this?

bubonicbob, 2024-01-15:
> The feature is included in the [v0.5.1](https://github.com/jrester/tesla_powerwall/releases/tag/v0.5.1) release. The feature can now be added to home-assistant. @bubonicbob Do you want to take care of this?

Not sure I'm qualified to add any new entities at this point.   :/   I wish I had a powerwall so I could see how it actually worked.   My hope was to get a [full snapshot](https://github.com/vloschiavo/powerwall2/tree/master/samples) of this multi-powerwall setups so that I could orient myself in the overall data schema.   Everybody is already so familiar with their own setup and I am working from zero experience with the powerwall.

ejbatts, 2024-01-15:
@bubonicbob can I help by supplying any outputs?

bubonicbob, 2024-01-15:
> @bubonicbob can I help by supplying any output data?

I'd really like to see the output of [this script](https://github.com/vloschiavo/powerwall2/blob/master/scripts/getsamples.sh) on your system if you are willing to try it out.   It basically runs every known query we can perform, appears to scrub out personal data, and saves it to a collection of files.    I'd love to see all that info.

Also I would need to see it actually working in the current HA interface.   I don't really know what it even looks like.   :)   I started working on it because I have a Tesla and I thought it looked cool.

ejbatts, 2024-01-16:
I need help to run this, I have no idea. But always willing to try...zoom?

bubonicbob, 2024-01-16:
> ct_voltage_references

Cool!   I suppose you can just delete the 2 lines from the script that are making these error calls so that it completes.   Like the lines containing `synchrometer/ct_voltage_references/options` and `system/networks/ping_test`.    Perhaps your newer firmware has disabled those calls.

I'm going on vacation for a the rest of the week so I won't be online too much, but I would really like to see those results.   What time zone are you in?   I'm in the pacific time zone.

ejbatts, 2024-01-16:
I am in Sydney AUS UTC+11. I have the dumps, how can I send these privately as I am exposing serial numbers etc?

bubonicbob, 2024-01-16:
> Yes it has written the files , how can I send these privately as I am exposing serial numbers etc?
> 
> I am in Sydney AUS UTC+11.

Thank you!!!

I think this dropbox link will work to upload your files.   I set it up for upload only.

https://www.dropbox.com/request/SnMyhxNgVWTGoE8I8zwE

ejbatts, 2024-01-16:
@bubonicbob Files are uploaded.

The data I am wanting to access is shown in the file `running/system_status.json` .

The same file in the `stopped` folder returned an error, and is titled `system_status.err` .

bubonicbob, 2024-01-17:
> @bubonicbob Files are uploaded.
> 
> The data I am wanting to access is shown in the file `running/system_status.json` .
> 
> The same file in the `stopped` folder returned an error, and is titled `system_status.err` .
> 
> I think all serials are hidden so feel free to share the files with developers.

Thank you!!   I'll take a few days to look it over.    I hear so many cool things about the outback culture down there...   I can only imagine what a cool setup you have.

bubonicbob, 2024-01-18:
> * Powerwall State (eg Grid Following)

Hey @ejbatts , when your powerwall batteries are full and in grid following state, how many watts does a Tesla app/web show your powerwalls are consuming?   It is typically like 150 watts or more like 2100 watts?    I'm curious what an official Tesla app/web page shows.

I'm really wondering if they show you "real" or "apparent" power.   Seems like you would want to know the actual power flowing into or out of your batteries.

ejbatts, 2024-01-19:
**I have uploaded my email address.**

>  how many watts does a Tesla app/web show your powerwalls are consuming? It is typically like 150 watts or more like 2100 watts? 

Typically about 150W. **I have** **uploaded a graph** for a few hours yesterday at 100% SOC. Tesla numbers are all rounded to one decimal place so the app usually displays -0.1 or -0.2kW when charge is 100% and I am exporting excess solar to the grid.

> Also which parts of your system are running ~420 volts? I don't know what "load" and "site" are measuring

I think ~420V is DC voltage. Load = HOME consumption. Site = GRID import/export.

> I'm really wondering if they show you "real" or "apparent" power. Seems like you would want to know the actual power flowing into or out of your batteries.

I feel you may be over-thinking my feature update request.  The Tesla Integration in HA core already pulls data from several of the API endpoints. **I have uploaded a screenshot** of the data in my HA (V2 shows the hidden entities as well). The endpoints in my system are only related to the Powerwalls, but I presume there are more if a vehicle and charger are added.
To be clear, the data I would like to expose exists in the 'system_status' endpoint, because this is where individual battery data is presented as opposed to aggregated data that Tesla uses on its own app and web interfaces. There are 21 sets of data for each of my batteries. I would like to expose 9 of these as per below. The MOST important one of these to me is "p_out",  being the instantaneous measure of actual power flow in/out of the battery. The second most important are the four energy values. The least important are the electrical values of q,v,i and f. Here at least you can see our nominal supply parameters of 245V/50Hz (v/f)
 
>      "nominal_energy_remaining": 15137,
>       "nominal_full_pack_energy": 15137,
>       "p_out": -100,
>       "q_out": -1090,
>       "v_out": 245.60000000000002,
>       "f_out": 50.037,
>       "i_out": 0.1,
>       "energy_charged": 610483,
>       "energy_discharged": 509907,
>

PR #60 (2024-01-15): feat #57: Expose more low level information of discrete batteries
closes #57