jrester/tesla_powerwall #70: Log in fails with error "no running event loop"

ukoda, 2025-01-29:
I have been using an earlier version of tesla_powerwall for a few years now with no trouble.  After a server hardware failure I have built a new Rocky 9 server, used pip to install the latest release of tesla_powerwall and I trying to get my code back up and running.  It is throwing an error from a library it is calling:
```
[root@ichi ema]# ./emacollectpowerwall.py 
/usr/lib/python3.9/site-packages/requests/__init__.py:87: RequestsDependencyWarning: urllib3 (2.3.0) or chardet (4.0.0) doesn't match a supported version!
  warnings.warn("urllib3 ({}) or chardet ({}) doesn't match a supported "
[root@ichi ema]# ./emacollectpowerwall.py 
^CTraceback (most recent call last):
  File "/opt/ema/./emacollectpowerwall.py", line 32, in <module>
    powerwall = Powerwall("10.20.1.160")
  File "/usr/local/lib/python3.9/site-packages/tesla_powerwall/powerwall.py", line 30, in __init__
    self._api = API(
  File "/usr/local/lib/python3.9/site-packages/tesla_powerwall/api.py", line 36, in __init__
    jar = aiohttp.CookieJar(unsafe=True)
  File "/usr/local/lib64/python3.9/site-packages/aiohttp/cookiejar.py", line 95, in __init__
    super().__init__(loop=loop)
  File "/usr/local/lib64/python3.9/site-packages/aiohttp/abc.py", line 177, in __init__
    self._loop = loop or asyncio.get_running_loop()
RuntimeError: no running event loop

During handling of the above exception, another exception occurred:
```
Maybe there is a missing module I have not installed?  Reviewing the latest sample code I think I am doing things right.

My code, pruned and person info modified, is:
```
from tesla_powerwall import Powerwall
try:
  powerwall = Powerwall("10.20.1.160")
  powerwall.login("PASSWORD", "USER@DOMAIN")
except Exception as e:
  logging.error(f'Failed to log in to Powerwall {e}, waiting 1 minute, then exiting to restart')
  time.sleep(60)
  exit(1)
```
Any suggestions welcomed.  Pleas let me know if additional information is need,

Thanks
  David

jrester, 2025-01-30:
Hey @ukoda, since release [0.5.0](https://github.com/jrester/tesla_powerwall/releases/tag/v0.5.0) this library is async and therefore requires a running event loop. The easist way to get an event loop is the [`asyncio` ](https://docs.python.org/3/library/asyncio.html) module from the python stdlib. Then your code would have to look something like this:

```python
from tesla_powerwall import Powerwall
import asyncio

async def main()
  try:
    powerwall = Powerwall("10.20.1.160")
    await powerwall.login("PASSWORD", "USER@DOMAIN")
  except Exception as e:
    logging.error(f'Failed to log in to Powerwall {e}, waiting 1 minute, then exiting to restart')
    asyncio.sleep(60)
    exit(1)

asyncio.run(main())
```

You can also take a look at the [`example`](https://github.com/jrester/tesla_powerwall/blob/master/examples/example.py) which showcases some more features with async.

ukoda, 2025-01-30:
@jrester Thanks the quick reply.

As a sanity check I ran the example program and it looks good.

I made those changes and it does get further, crashing because:
```
    Solar_power =  convert_to_kw(meters.solar.instant_power,   precision=3)
AttributeError: 'coroutine' object has no attribute 'solar'
```
I'm guessing some the structures have changed, so I will need take the time to go thru my code and workout what needs reworked.

I think we can close this issue since the problem is simply my not being aware of the changes in this module.

Keep up the good work :-)

jrester, 2025-01-30:
Alright, good to hear that it is working again :)

closed 2025-01-30 by jrester